unkyulee/micro-journal #2: Mounting SPIFFS failed

CelloGit, 2024-04-29:
Hi Unkulee,

Being an avid fan of writerdecks, I felt fortunate to stumble upon you Micro-Journal project on GitHub. Great stuff.
Feeling lazy for a change, I just missed purchasing your build of the fifth iteration on Tindy (the shopping cart was emptied while I was preparing the payment), which was all the better because building these contraptions is really as much fun as using them.

But alas, I am stuck and wonder if you could give me a clue how to proceed.

All went fine until I Upload and Monitor micro-journal-rev-4-esp32. Uploading goes well but then the TFT SPI Screen gives error message "SD CARD MISSING" and the terminal in VS PlatformIO gives:

E (9) SPIFFS: mount failed, -10025
[   223][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
SPIFFS mount failed!

Config Setup
----------------------
Loading config ...
App Status loaded
Opening config.json file
[   479][E][vfs_api.cpp:24] open(): File system is not mounted
config.json file doens't exist
...
Init Keyboard
US
Keyboard layout changed null
null

I tried two different TFT Screens both tested for display and SD functionality, the wiring has been checked 4 times and the ESP32 S3 behaves well in other situations.

PS: Back and Menu buttons are wired to pin 16 and 15 respectively, but where do the other pins go to?
PS2: Build Guide Step 3. Wiring to ESP32: PIN 11 says MOSI - SD but from your excellent descriptions I gather this should be MOSI - SD, DISPLAY. Correct?

Thank you very much,
CelloGit

unkyulee, 2024-04-29:
[image]

unkyulee, 2024-04-29:
PS: Back and Menu buttons are wired to pin 16 and 15 respectively, but where do the other pins go to?

Back button should be pin 16, and other pin to GND
Menu button should pin 15 and other pin to GND

PS2: Build Guide Step 3. Wiring to ESP32: PIN 11 says MOSI - SD but from your excellent descriptions I gather this should be MOSI - SD, DISPLAY. Correct?

That is correct. I will update the guide as soon as possible. 

Thanks for the feedback!
and good luck with the build

CelloGit, 2024-04-30:
Solves it. Thanks. And nice wakeup.gif and sleep.gif!

closed 2024-04-30 by CelloGit