unkyulee/micro-journal #7: Doesn't boot on first power up

byzkarl, 2024-07-05:
When the device is powered off, flipping the switch will bring a white screen, but it won't go any further. Turning it off and on will resolve and it will boot. After this if I turn it off and on it will function correctly. 

It seems similar to this issue?

https://esp32.com/viewtopic.php?t=31123

I have the rev. 6 version running latest firmware (13).

unkyulee, 2024-07-05:
I think you are right about the issue.
I will try with gpio 0 pullup resistor and see if it resolved the issue.

Thanks for the feedback

Un Kyu lee

byzkarl, 2024-07-05:
If you do a repair, could you take lots of pictures? I'm not handy with a soldering iron. 

Gadget is very nice and I love it.

unkyulee, 2024-07-05:
[image: image.png]

Just trying to collect some information. When you open the front panel, and
see the ESP32 module. Do you see this white box thingy on the board?

byzkarl, 2024-07-06:
I'm sorry, but the image didn't come through. I will look tomorrow to see if there is a white box thingy.

unkyulee, 2024-07-06:
ESP32 seems to be going into Bootloader mode, which won't load the firmware
to users and just shows a white screen.
It seems to be consistently happening when the device has been turned off
for a while and turned back on.
For now, work around for this issue is to turn it off and turn it back on
again.

It seems that this is related to the ESP32 device, and some of the boards
are having this issue but not all the boards are having this issue.
I am not sure what is causing it and I am not equipped with a proper tool
to verify on the hardware level.
I will need more data to verify what specific model is having this issue,
or which supplier's board should work.

I will release the DIY build guide soon in the future.
With this guide it should be possible to replace the ESP32 module to
another type that is known to work properly.
Soldering will be required in this case...

It's really unfortunate.. and I don't have capacity to anticipate this
level of hardware issues.
I understand that it is not the perfect solution, but at the moment, the
best is to turn it off and on again twice at the very beginning of the
power on.
I will gather more information, and follow up with this issue as more
findings come out.

Un Kyu Lee

unkyulee, 2024-07-06:
[image]

I have found a way to resolve this issue of not boot up at the first start up. 
This requires soldering.... 

Here is the remedy. 
Pull out ESP32 module from the board. There should be glue around the board. Try to use some sharp pinset to pinch out the glue without breaking the board or the wires. Once the majority of the glues are scooped out. Carefully leverage the board from the bottom to remove it from the enclosure. 

Once it is removed. Connect the 3v3 and GPIO 0 pins with 10k~ ohm Resistor.  My recommendation is higher than 10k. Reason being if the resistor value is too low, it can flow too much current and break the core. Or... it will consume rather significant battery if the resistor value is too low.

Once the connections is made such that GPIO 0 is always going to be 3v3, the issues with the first power up issue seems to be not happening any more.

byzkarl, 2024-07-10:
The repair has worked for me, even with my messy soldering.

zuperdzigh, 2024-07-28:
oh no, is it possible that's what happening with mine?  the screen suddenly went white.  And I can't recover it by turning it off and turning it on again.

unkyulee, 2024-07-28:
Most likely batter is low
Un Kyu Lee

zuperdzigh, 2024-07-29:
it is still white. I tried changing the battery and tried to charge it.

unkyulee, 2024-07-29:
https://github.com/unkyulee/micro-journal/blob/main/micro-journal-rev-6-one-piece/build-guide.md#flashing-the-firmware-for-the-first-time

Can you try to upload the firmware to esp32?

Un Kyu Lee

zuperdzigh, 2024-07-29:
checked the cables and then tried to flash the firmware,
received this error message:
A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.

unkyulee, 2024-07-29:
I have a bad feeling that maybe ESP32 board is broken. Let's hope that it's not the case. 

You may need to install a driver in order to recognize the ESP32 board. 
https://randomnerdtutorials.com/install-esp32-esp8266-usb-drivers-cp210x-windows/

try to install the driver and see if you can flash the firmware

zuperdzigh, 2024-07-29:
by the way, there is a solder residue on the RX pin, but I didn't see any wire to connect it to?

I have extra ESP32 board just in case, but I will try with the current board.

unkyulee, 2024-07-29:
Here is the build guide where it should have all the information necessary to build from scratch. 
You can refer to wiring table to understand purpose of each pins and where to connect to 

https://github.com/unkyulee/micro-journal/blob/main/micro-journal-rev-6-one-piece/build-guide.md

unkyulee, 2024-07-29:
You don't have to build from scratch. I wanted to say that the document should cover all the information that is needed to identify the problem

tektite-kz, 2024-07-30:
Hi, 

I hope it's okay to comment here, as I seem to have a similar issue. I've the USB-host version from scratch to give it a try, and I've encountered this issue where the board won't load after the initial firmware write. It starts to initiate the screen, and then bootloops. The screen is flashing white. I've already added an 11k resistor between 3v3 and GPIO-0, nothing changed. Any idea what I've done wrong? Could it be low power, because I have booted the board using the COM port connected to my PC? (Battery hasn't arrived yet.) 

Output from VSCode below:

ELF file SHA256: 7555f236856244cd

E (1362) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
��%N(��N!�H�L����␌�Jʳ̷%���
Device Started. Baud rate: 9600
[287] SPIFFS mount succeeded!
[288] SD Device CS: 9
[296] SD Device Initialized
[297] SD Card detected
[298] Loading config ...
[298] App Status loaded
[420] Opening config.json file
[424] Deserializing config.json file
[424] Loading app status config
[424] Config loaded successfully!
[546] Total heap: 321900
[547] Free heap: 291164
[547] Total PSRAM: 8386247
[547] Free PSRAM: 8359067
[669] Init USB Keyboard
[700] Setting keyboard layout: US
[700] DISPLAY SETUP
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4200df17  PS      : 0x00060630  A0      : 0x8200e008  A1      : 0x3fcebb80  
A2      : 0x00000010  A3      : 0x00000000  A4      : 0x60004000  A5      : 0x0000000b  
A6      : 0x000000ff  A7      : 0x00000008  A8      : 0x08000000  A9      : 0x3fcebb50  
A10     : 0x3fca327c  A11     : 0x00000001  A12     : 0xffffffff  A13     : 0x00000008  
A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000010  LBEG    : 0x42019b0c  LEND    : 0x42019b70  LCOUNT  : 0x00000000  

Backtrace: 0x4200df14:0x3fcebb80 0x4200e005:0x3fcebbb0 0x4200e010:0x3fcebbd0 0x4200aca0:0x3fcebbf0 0x4200c6c2:0x3fcebc20 0x4201b052:0x3fcebc50 

Any advice gratefully received, and thank you for creating such an inspirational product.

unkyulee, 2024-07-30:
It is possible that the source code may not be the latest version. Try to pull the latest code and see if any changes.

zuperdzigh, 2024-07-30:
i finally replaced the ESP32, now the VScode works but still white screen.  I already check all the wires and the battery.  is there any way to know if somehow I didn't connect the wire correctly, now that it is covered with solder?

unkyulee, 2024-07-30:
After replacing esp32 and display still coming up as white could mean that
wires on the display may be a problem or problem on the display itself. But
since the display comes up white, means that display itself should be ok,
some wires may not be connected well, data signals are not received well
from the display.

Normally, I check with eyes to see if any shorts and then do continuity
check with multimeter.

Un Kyu Lee

tektite-kz, 2024-07-31:
Thank you for replying,

I've checked my connections with a multimeter, and double-checked them against the wiring diagram. I have tried both 1.0.16 and 1.0.15, as well as cloning directly from GIT into VScode. I assume that I have either a bad ESP or the wrong display. The display has the same layout and specs as listed in the build guide, but has v1.2 written at the bottom instead of v1.1. The board is flashing successfully, both filesystem and program, so I'm going to order a different display and try that. 

best wishes,
Kz

tektite-kz, 2024-08-02:
New screen, no change. Wiring checked thoroughly (ever been so paranoid that you test premade DuPont connections?) so I'm going to assume there's either a fault with this ESP32 or my build environment. Sorry for cluttering up your bug report, I thought it was the same issue.

zuperdzigh, 2024-08-03:
> New screen, no change. Wiring checked thoroughly (ever been so paranoid that you test premade DuPont connections?) so I'm going to assume there's either a fault with this ESP32 or my build environment. Sorry for cluttering up your bug report, I thought it was the same issue.

Hi, just wondering, are you using windows or mac or linux?  I also tried wiring, rewiring, changing display, changing ESP32, the only thing that I haven't try is using platformIO on windows.  Does it matter?

tektite-kz, 2024-08-03:
Hi, that's a really good question. I'm using Windows 10. I did try flashing
on Linux, but there was a dependency for the screen that refused to
resolve. Annoyingly, on Linux I had to use the pio command line to install
dependencies too, whereas on Windows that happens automatically. I am new
to VSCode though, so maybe I've missed a trick.

anachronistick, 2024-09-02:
> New screen, no change. Wiring checked thoroughly (ever been so paranoid that you test premade DuPont connections?) so I'm going to assume there's either a fault with this ESP32 or my build environment. Sorry for cluttering up your bug report, I thought it was the same issue.

Sorry to further clutter up this bug report, but when I was building my rev.5 I also had trouble getting the screen to work. Looking through the TFT_eSPI user file examples, there is one listed for esp32 s3 and it says to use only pins 0-31 for TFT_DC and TFT_RST. I remapped TFT_DC from 46 to 8 (moved wire accordingly) then the display worked fine.

tektite-kz, 2024-09-15:
Hi,

Sorry for not replying sooner - work has been busy.

It's great that you got it working - nothing like solving a problem! I used
the pins in the original instructions without issue. Perhaps you had an odd
board, or a cold solder joint.

Best wishes,

Kz