unkyulee/micro-journal #10: [Feature request] Alternative word delete keybinding?

benhoh, 2024-07-20:
Could you include an option to map `alt-backspace` to the delete word function, rather than double backspace? This would make it similar to Freewrite and Pomera devices. 

My muscle memory is to hit backspace multiple times if I realise I’ve made a typo right after typing a word. With the current editor behaviour, I always end up deleting multiple words accidentally. 

I hope it’s okay to make feature requests here. Let me know if there’s a more appropriate place. 

Also, a question about your roadmap: is your plan to follow the old Freewrite “strictly no editing” approach, or will you be aiming for document navigation and cut/copy/paste? I respect both approaches — just curious.

unkyulee, 2024-07-21:
I have recently made an update for alt delete, and also multiple files.

Can't really say that I have a roadmap. Some communications are done in
writerDeck subreddit.

https://www.reddit.com/r/t5_6exk8q/s/k8ea36zkZE

Trying to add cursor navigation and some edit features in the future.
Un Kyu Lee

benhoh, 2024-07-22:
I was outpaced by your rapid development! It’s really taking shape quickly, and the multiple files are a godsend.