unkyulee/micro-journal #11: Installed 1.0.15. Something odd happened.

Gabey8, 2024-07-22:
I’m using file folders for the first time. 

Once I hit a certain point (journaling about my weekend) , it didn’t seem to want to let me type. I also noticed huge jumps in file size every time I backspaced a few times.

When I synced the file, I saw that the file’s contents were tacked onto the end of the file multiple times. 

I can email the file if you want to see it. 

Is there a file size limit, above which we should sync and start a new file?

unkyulee, 2024-07-22:
This must be a bug, could be that some edge cases are not handled
correctly. I will check the save procedures and see if this can be caught.
Sorry for the inconveniences
Un Kyu Lee

cassidoo, 2024-07-24:
Hi! I just ran into this exact problem as well, and caught it on video!

https://stream.new/v/uSya91xc8nwbLY5cd4fZbMfkysinw2luOwIJHbVhQT4

cassidoo, 2024-07-24:
I will download the new firmware! For what it's worth, in this case, I noticed it started when I typed a capital letter I? Like it jumped a little bit when I did! Anyway, I'll try the new one!

Gabey8, 2024-07-29:
With the most current firmware (version 16 r3), I was able to create a file that was over 11,000 bytes with my Rev.6, and I didn't run into this issue. I also haven't noticed the file sizes suddenly being incremented more than expected. 

Is there a file size limit, or can we create files of any size if the SD card has the space for them?

unkyulee, 2024-07-29:
File size limit is 4gb

Un Kyu Lee

Gabey8, 2024-07-30:
Fabulous! Not even I am verbose enough to exceed a 4 GB limit, lol! :)