unkyulee/micro-journal #13: [Feature request] Scroll up and down?

cassidoo, 2024-07-24:
In a couple writing sessions now, I've written a bunch, and then walked away and come back, and forgot all the points that I touched on when I wrote. Would it be possible to add a keybinding just to scroll up and down so I can quickly remember and get back to writing? I don't need to fully edit earlier lines (because I think that's not the point of the device!), just scroll to look, and then move on.

unkyulee, 2024-07-24:
Cursor navigation is currently in work. Version 16 is going to have it
implemented. Right now it is very rough and unstable. But if you want to
try it out. There is dev version in the release page. Make sure to backup
you SD card before heading into dev version.
Un Kyu Lee

cassidoo, 2024-07-24:
Awesome, thank you so much!

Gabey8, 2024-07-24:
I love the latest update! I was able to get the arrow keys to work on my Rev.5 without difficulty.

However, I have a smaller-sized wireless keyboard, which uses a function key to access Home (which I know isn't activated in this revision), PgUp, PgDn, and End. Whenever I hit the Fn key, I get a slash character on the screen, so I can't test page up, page down, or End.

With the Rev.6, I could only get it to update if I renamed the file to firmware.bin. Once I did that, I went into the keyboard.json file. I made a copy of it (as I have a customized "space after B" version of the file), then gave myself arrow keys on the WASD keys. Under that, I put HOME (for when it gets activated later, which I suspect is on the to-do list), PGUP, PGDN, and END on the Z, X, C, and V keys.

PGUP and PGDN worked. But END sent me to the end of the text line, rather than to the end of the document.

Oh, and interestingly, when I updated to Rev 15, the number of the file to choose wasn't initialized yet. So I have File 0 on both my SD cards. :) This means that in a future revision, maybe we can have ten files to choose from if we include a File 0, lol. Hey, the more, the merrier! We all like to write a lot, after all.

Thanks for being so responsive with all these updates!

unkyulee, 2024-07-25:
Thanks for the feedback.

For rev.5 issues for various keyboards are really difficult to anticipate.
I would imagine fn key not to be sending any signals to the connected ends,
but it seems like it does. And the firmware is trying to map to the keys.
This is something I can debug when I have that specific keyboard.

Hmm... File 0 issue should be checked. It might lose texts when file
numbers are not set. But maybe could be just another file slot can be
added. Hmmm... This is quite a bug here. Thanks for reporting it.

I have decided to make end key going to the end of the screen line rather
than end of the file. Felt like a more intuitive approach. Suddenly going
back to the end of the file could be, a bit too sudden? Many details can be
adjusted as time goes.

I am very happy that cursor navigation has reached a point, where it feels
very much viable and smooth. Please, keep feeding me with your feedback.

Thanks!

Un Kyu Lee

unkyulee, 2024-07-25:
File 0 issues are resolved. Now total of 10 file slots are available.

Gabey8, 2024-07-25:
The best bugs are the ones that can be polished up and converted into features! 🙂👍🏻