unkyulee/micro-journal #16: Display fails to initialise after flashing

tektite-kz, 2024-08-04:
Hi, 

Opening a new thread as this is clearly unrelated to the issue with GPIO-0 and booting. I've tried two screens, and two different ESP32-S3 boards, several different revisions of code, and finally a fresh install of VS/Pio: I cannot get past this error. The code builds fine, and flashes without complaint. Then the monitor shows the board bootlooping and the display shows white, flickering slightly with each bootloop. The backtrace states the following:

0x4200f678: TFT_eSPI::begin_tft_write() at TFT_eSPI.cpp:81
 (inlined by) TFT_eSPI::writecommand(unsigned char) at TFT_eSPI.cpp:982
0x4200f769: TFT_eSPI::init(unsigned char) at TFT_eSPI.cpp:692
0x4200f774: TFT_eSPI::begin(unsigned char) at TFT_eSPI.cpp:603
0x4200b180: display_setup() at display.cpp:26
0x4200d56a: setup() at main.cpp:30
0x4201dc1e: loopTask(void*) at main.cpp:42

Looking at the earlier thread (https://github.com/unkyulee/micro-journal/issues/7) it looks like another user has a similar issue.

Any ideas what I've done wrong, or is this a bug with the display library? 

Best wishes,
kz

tektite-kz, 2024-08-14:
I've figured this out, and it can be closed. This behaviour is caused by an issue with the TFT library, and the solution is here: [https://github.com/Bodmer/TFT_eSPI/issues/3304#issuecomment-2095080442](url)

Tested, and now working. I changed line 2 in platformio.ini to this: 

`platform = espressif32 @ 6.5.0`

unkyulee, 2024-08-20:
Thanks for reporting the problem and also recording the solution. I have applied to the project configuration as you have specified.

I think leaving the issues open is fine. If in any case when someone encounters similar problem, it can be helpful.

tektite-kz, 2024-08-20:
No worries! I'm just pleased to have contributed something, even in this tiny way. This is such a cool project.