unkyulee/micro-journal #18: File wiped turning v6 on and off

HookPro1, 2024-08-09:
Can't tell what actually did this and it must be rare because I have been using the v6 for some time.  I wrote for a little over an hour. I had to get up to do something quickly and I turned the v6 off. (I have done this before). When I returned and turned the v6 on the filespace I had been working on (which it booted into) was blank and showed 0 bytes.  I checked the file on the card and it was empty as well. 

Obviously, the safest protocol (and what I should have done) would be to sync every time you are  stopping using the v6, then turn it off. 

So, if an electrical event happened when turning it off affect the file on the card?  Because it was still open?  I really thought the fact that it was saving all along would rescue most of my writing, but it didn't.  

So, was this likely an electrical event?  I will keep testing, just not with writing I care about.  ;-)

HookPro1, 2024-08-10:
I've done an extensive amount of testing trying to reproduce this and I can't.  I'm no longer convinced it amounts to an issue, more like just a strange event that is unlikely to happen.  I do know, after much testing, that if you shut off the v6 before the red light goes out, you will lose what you just typed (not surprising).  But I never lost everything in the buffer and it never wiped the file on the SD card.  But If I go to the menu while the light is red, the save is completed and I lose nothing.  So clearly, that is the protocol (it should have been obvious to me) to be pretty guaranteed nothing bad happens.  So, disregard this for now and hopefully it won't come up again.  I'm confident enough to go back to drafting on the v6.

unkyulee, 2024-08-20:
I am really sorry for the issue. It is critical that text contents are not lost. I am taking this issues very seriously, and putting all my effort to prevent the loss as much as possible. Please, when you find any other cases or more clues what may have caused it, let me know ASAP. 

For your information, I would like to clarify the cases when files are saved, so that you have more clear picture how to prevent the further cases. 

# 1
When idle for 5 seconds. When there are something typed on the screen and there has been no changes for 5 seconds. Meaning no keyboard typing for that amount of time. Auto save will occur.

# 2
When entering menu. Pressing menu will save the file immediately before entering the menu. In order to have the latest text saved on the file for the sync process.

# 3
When the screen buffer is completed. There should be around (at the moment of the comment) 2,000 ~ 4,000 bytes worth of buffer. When the buffer is full, then it will save the text to the file and empty the buffer.