unkyulee/micro-journal #25: New keyboard bricked my v5?

darryllee, 2024-09-11:
I bought a Royal Kluge F68 to use with my v5. Plugged it in, and the indicator light that says glows red when connected to my computer, flickered off/on while plugged into the v5.

After a few restarts and replugs, keyboard seemed to "settle" down, and I typed a bit.

Turned it off and back on again, and now device keeps reverting to a flashing white screen, whether an SD card is plugged in or not. Is my firmware hosed? Should I try a refresh?

darryllee, 2024-09-11:
Oh crap.
> Connecting directly to the keyboard that requries charging will break the Micro Journal
*Sigh*. Is there any fix?

darryllee, 2024-09-11:
So I left the device off for a while. I've put the new firmware on my card, and when I turn it on, I briefly saw the message asking if I wanted to update firmware, but it then immediately started blinking white again.

Did I burn up my ESP32? :_{

unkyulee, 2024-09-11:
Charge the battery. Blinking white could mean the battery is low
Un Kyu Lee

darryllee, 2024-09-11:
Whew, ok - charged overnight, seemed better this morning, but after trying a different keyboard (that used a dongle) it also went white. Swapped out the battery completely and it was much better. Top port is for charging, right?

I'm thinking about adding a battery indicator for mine like you did for the Rev.6. It'd be nice if it (or something similar) could also show if it is charging... I wonder if there's something else that does that.

unkyulee, 2024-09-11:
I think, easier solution to measure the battery is the what has been for Rev.6 
or a lipo battery charger can be used to see if the battery is fully charged and charge it outside before using it, just to make sure. 
If you have a voltage meter, you can use it to measure the charge level as well. Should be bettwen 3.4 V ~ 4.2 V - 4.2 being full.

darryllee, 2024-09-13:
Hi - so yeah, the battery that was in the device while I thought it was charging was actually only measuring 2.7V. AH, my charger is 3A, and I see that the max charging current for the charging module is 1A. I'll try to find a smaller charger.

darryllee, 2024-09-17:
It's interesting. So this keyboard is now working ok, now that the battery of the ver.5 is fully charged. I can type in any of the slots, look at various settings, etc.

But when I try to Sync, if the backlight on the keyboard is on, it seems to draw too much power, and the screen starts flickering white again. I wonder if my problem is that this is only a 2500mAH battery?

darryllee, 2024-09-17:
I guess an indication that it's a power issue is that when I plug in a charging cable into the ver.5 (top port) and then try syncing with full backlight, there are no issues.

unkyulee, 2024-09-17:
When syncing, it uses wifi and it can draw a lot of power and if the keyboard is also drawing a lot of power then it can over burden the power supply and shuts down. I would recommend to turn off the led, or remove the keyboard... Hmm removing may not be possible as you'd have click the button.... Hmmm ....