unkyulee/micro-journal #26: Rev. 6 - Text Display Truncates after syncing more than 2000 bytes

anon3792737, 2024-09-25:
I've noticed that after writing more than roughly 2,000 bytes or so, and then syncing the file to Google drive, the display of the text truncates on my microjournal.

 I'm not exactly sure if it's at 2,000 bytes, but it seems to be around that general size.. 

The text isn't actually removed from the file.  For example, if I keep typing and rsync to Google Drive, all of the original text is still there. However, I can no longer navigate back to the text that I saw before the display truncation.

unkyulee, 2024-09-26:
It expected behaviour. Every 4000 characters (bytes), saves to the SD card and then clears the buffer. 

In that case it's likely that previous buffer won't be able to be navigated. 
With the recent introduction of cursor navigation, should be possible to go back to the previous buffer. 
To be implemented in the future.

unkyulee, 2024-09-26:
Don't close it. I think it's better that some people with same problem might find this conversation useful.