unkyulee/micro-journal #27: Can I add Korean Language to this writedeck?

abubaman, 2024-09-27:
I am curios about the possibility to add a Korean language to this device.
If it doesn't have this funcionality, when do you expect to add Korean language to it?

unkyulee, 2024-09-30:
Rev.2 should be able to support Korean language. It is one of the motivation of building rev.2 is to cover Korean. I will test in the near future and post it on my youtube channel. 

https://www.youtube.com/@unkyulee

lilong10, 2025-04-24:
Is Chinese input supported in Rev 7?
Thanks!

unkyulee, 2025-04-24:
Rev.5, 6, 7 - No, it does not support Chinese language.

Rev.2 - Yes, it does support Chinese language input

lilong10, 2025-04-24:
Cool! Any plan to support it in Rev 7?

kolyan94, 2025-07-31:
Is it possible to add Cyrillic chars?