unkyulee/micro-journal #28: Recommended adhesive for mounting circuit board to 3D prints

Kingseso, 2024-10-03:
Hello,

I'm building the micro journal Rev 6 and I was wondering what has worked well for others regarding mounting the ESP controller and battery charger to the 3D printed enclosure (super glue, hot glue, epoxy, etc.).  I'm worried that when the USB is constantly plugged in and out, that the boards sooner or later lift from the print.

Thanks in advance,

Cecil

unkyulee, 2024-10-03:
I use the engineers best friend, glue gun.

themoderndance, 2024-10-03:
Maybe this doesn't apply, I haven't looked at the design yet but I prefer to use threaded inserts and screws in all my projects when possible, or at the very least embed nuts in the model so I can screw into those.

Kingseso, 2024-10-03:
> I use the engineers best friend, glue gun.

I'll give the glue gun a go.  I'll try to find a heavier duty glue gun.  I have a simple one for arts and crafts.

Thanks

Kingseso, 2024-10-03:
> Maybe this doesn't apply, I haven't looked at the design yet but I prefer to use threaded inserts and screws in all my projects when possible, or at the very least embed nuts in the model so I can screw into those.

I agree, but the ESP and battery charger have no mounting holes.

themoderndance, 2024-10-03:
Do you 3D print? In those situations I would design around that. I'm at work now so I can't look up what the inside of the design is like at the moment.