unkyulee/micro-journal #32: German Keyboard Layout Does Not Work in 1.0.17 Rev.5

looking-eye, 2024-10-31:
I upgraded to Rev. 5 today and now my German (GE in the menu) layout behaves exactly like US. Other layouts (Belgian for example) work as expected.

unkyulee, 2024-10-31:
Sorry for the troubles. I think new version has some migration issues. I
will test theg german layout and update them soon.
Will leave a comment here when it is updated.
Thanks for the report.

For the time being please use the previous version....

unkyulee, 2024-11-01:
This issue should be fixed now. I forgot to add condition statement to load German layout. Here is the link where you can download the latest releases.

https://github.com/unkyulee/micro-journal/releases

Please, note that, German layout (or other layouts) are not perfect. When you discover missing keys, or some improvements. Please, let me know.

looking-eye, 2024-11-04:
I can confirm, this works! Thanks!

closed 2024-11-07 by looking-eye