unkyulee/micro-journal #35: Can't Connect to Wifi

Miiraaj, 2024-12-01:
Hi, 

I got the Micro Journal, and I have followed the instructions to setup Wi-Fi, but every time I try, I get the following error message:

_There was an error running option S1 Wireless LAN_

I tried connecting to both 2.4GHz and 5GHz connections, but got the same both times. 

How can I fix this, or is there a way to get the files off it locally? I looked at the SD card on my computer, but couldn't see any of the Word Grinder files. 

Thanks for any help!

unkyulee, 2024-12-02:
Thanks to Hollis there are community solution to approach to this issue. Please, try the following instructions mentioned in the link below.

https://gist.github.com/hollispotter/c93439e79e13f8828ddcbe0043a3ecd8

Miiraaj, 2024-12-04:
Thanks for the link!
So I looked into this and attempted to do it, but I'm not exactly sure how to do it. 
I can find the etc folder, and a  modprobe.d folder, but there is no brcmfmac.conf folder or file to copy the lines of code into (see screenshot below for what I have). Do I have to create the file? If so, how do I do that? I haven't used the system much, and I'm not used to a keyboard where so many keys are hidden, so I'm kind of lost. 
Thanks for any help!

 
 

https://gist.github.com/hollispotter/c93439e79e13f8828ddcbe0043a3ecd8