unkyulee/micro-journal #36: Keyboard Not Responding

gsmoss76, 2025-01-02:
New "Mother of Twins” model - got it last Friday. It was working great. I made a couple of modifications to the keyboard layout and swapped the key caps. Was fine for half an hour but now does not respond to typing or knobs. What steps can I take?

thanks!

unkyulee, 2025-01-02:
I would check if the keyboard controller works well by connecting to the
pc. Also control the battery if it is fully charged. Also reboot and see if
that changes.

If the problem can't be fixed by you, i can have a look in the condition
that you cover the shipping and material costs.

Would it be an option that you'd consider?
Un Kyu Lee

gsmoss76, 2025-01-02:
> Thanks - I appreciate your quick response. The keyboard does indeed work when connected to my pc. I am charging the battery to see if there's any improvement. I will follow up if nothing changes.
>
>
>
>
>
> best
>
>
> GM

Hello Mr. Lee:
  
​
  
​I let the battery charge overnight, but still cannot power on and use the keyboard. The keyboard isn't responding while the microjournal is tethered to the computer either. I thought I might return the keyboard to it's initial setup but Vial isn't connecting either.
  
  
​​
  
  
  
  
​***
  
​www.gregorymoss.com

gsmoss76, 2025-01-02:
FYI - this is a screenshot of the error message I get when trying to use Vial.
  
​
  
​thx
  
​Greg
  
​
  
​
  
​
  
  
  
​***
  
​www.gregorymoss.com

gsmoss76, 2025-01-04:
What address should I ship the unit to for repair?
  

  
thx
  
agreg

unkyulee, 2025-01-08:
Sorry to hear that the keyboard is malfunctioning... it could be that keyboard controller is broken and it may need a replacements. 
Please, send me a message in tindie so that I can share my address in the private message channel. 

Thanks for your understanding. 

Un Kyu Lee