unkyulee/micro-journal #38: Won’t turn on anymore

joelvs456, 2025-01-17:
The GPIO isn’t the issue here. Device worked fine at first, but recently, after charging and flipping the switch, it won’t turn on anymore—just a flash, then a blank screen. I’ve replaced the battery and tried a power bank, but nothing.

unkyulee, 2025-01-17:
Try to plug a usb cable to the ESP32 directly. Which is place on the left hand side. When power is supplied from this way, ESP32 should have no influence on the battery system and should be powered on directly. If this method gives a proper boot up, then it is highly likely that the battery is at fault. I would check the battery voltage and see if the voltage shows correct value. If the value is lower than 3V... then you may need to use other battery or use shock method to revive the battery. Not recommended this way. Try to purchase the battery from the trusted vendors. 

If this method still makes the same problem. 

Then it is possible to pin point the issue to ESP32 or the DISPLAY. Also, it is possible that some wires may have issues. I would open up and see if the wires has any visual impairements. Then also try all the possible power on with the display enclosure opened up. If none of the method can get to the correct boot up sequence... then most of the time it may require to replace the ESP32 module.

joelvs456, 2025-01-18:
I tried connecting to it on the right hand side same thing happened. All wires seem fine from what I opened it up. I tried to see if it was display. Since it’s been off but i do think it’s the esp 32.

unkyulee, 2025-01-18:
Most of the time, it was esp32 in my case. You may need to replace the module, if this is something you can do there is a build guide specifying the wiring table. 

https://github.com/unkyulee/micro-journal/blob/main/micro-journal-rev-6-one-piece/build-guide.md

If this is not available to you then I can have a look and repair them. Send me a via tindie and I will share you where to ship and so on. 

https://www.tindie.com/stores/unkyulee

unkyulee, 2025-08-07:
Sure. Send me a message in tindie and i will send you a shipping instructions.