unkyulee/micro-journal #39: Rev.6 Sync - failed

timeandspacemachine, 2025-02-07:
Loving the Rev. 6 so far — everything I wanted it to be :) But I can't seem to get the Google Drive sync working. Got connected to the internet fine, no error messages on the rev. 6 nor any emailed to me, and the rev. 6 says the syncs are successful. But in the 'executions' tab on the script on Drive it says the attempts have all failed at syncing. Any suggestions?

unkyulee, 2025-02-07:
Probably if you dive deeper into identifying the problem, maybe a solution can be found... but I am going to take a typical IT support fashion. Trying to start from the beginning. 

Delete the appscript on the drive and try to follow the steps one by one again. It's possible that one of the steps may have been skipped or set differently. I also make mistakes during the setup and just restarting the setup made it working... I can't say what was missing or what was wrong,

timeandspacemachine, 2025-02-07:
Redoing it worked! Probably the most obvious of solutions, alas tech troubleshooting is not my forte. Thanks so much, Un Kyu!