unkyulee/micro-journal #41: File wiped after random restart

DreaminDani, 2025-02-13:
I'm filing this issue on behalf of my parter who started using the micro-journal in August. Here is her report:

> I was typing and the screen flashed grey, just like when it's about to run out of battery.
> So I replaced the battery with another one, thinking that it was just dead.
> But when I turned it back on, the file I was typing it was completely blank.
> I tried to sync the empty file, thinking that maybe it was just a display issue and I could at least see the text contents on google drive. But then I ran into this issue: #19 

Plugging the SD card into a computer shows that the file is indeed wiped. I tried to use recovery software on it but there's trace of the previous version of the file. I have a feeling it is actually gone, but I wanted to report the issue in case it could help prevent future data wipes

unkyulee, 2025-02-13:
Oh... what a tragedy to lose the contents... ajajajaja. I don't know how to say and what to say in order to relieve the pain that your partner went through or going through now. 

Most likely the battery went low, and the device start to turn on and off continously. When this is happening for a longer duration it can wipe the SD card... it's even possible that SD card can get destroyed. It's something I noticed after quite some time later since the release. I have added some guide to add battery indicator to help out these issues but fundamentally, battery going critically low and the device will behave unexpectedly and some cases causes damage to the components. 

Unfortunately, I don't have a ways to systematically prevent this issues. However, I will make a note on the quick start guide about the low battery issues and make sure that the battery is charged up and not to leave it blinking for a longer duration. 

Again, I am very sorry for the issues that your partner went through, and I don't think any of what I said is helping.... but I hope it helps to clarify the current situation.

DreaminDani, 2025-02-13:
That's helpful clarification, thanks!

Can you explain how the battery works a bit more? Right now, she has to take off the back plate to swap the batteries out every time it dies. I was unaware of a battery indicator, but maybe that's something she can pay more attention to and swap out earlier.

Is there a way to charge the batteries without swapping out the batteries altogether? Maybe the best way to prevent data loss would be for her to keep it as charged as possible at all times.

She's using a rev.5.

unkyulee, 2025-02-13:
Top USB port is the charging port. It will charge the battery when plugged in.

unkyulee, 2025-02-14:
it's USB interface so voltage/ampere requirements are more or less controlled. Note that PD chargers won't work. Typically a cable with USB-c to c is not going to charge the device.

There should be a light dmming out through the port when it is red means charging, no red means, either no battery or fully charged

DreaminDani, 2025-02-14:
Thanks so much! We'll give it a shot.