unkyulee/micro-journal #47: Shift key not working.

Ryonne, 2025-03-02:
Hey! I ordered a Micro Journal rev. 5 and am overall very happy with my product. The only problem I'm dealing with right now is that the shift key isn't working, so I'm not able to make capital letters or certain punctuation marks. Is there any way to fix this, or is this how the Micro Journal is?

unkyulee, 2025-03-02:
Some keyboard has reported to be not working with shift keys. Would you share what keyboard you are using?

Ryonne, 2025-03-02:
Tested it out on two separate keyboards: ROG Strix Scope II 96 Wireless and the Keychron V1 Max Wireless, both using the wireless dongles rather than directly plugging them in because it overloaded the battery on direct connection in both cases.

What keyboard would you recommend me to use instead?

unkyulee, 2025-03-02:
If you have rather classic office keyboard just to test the features. Would help. Do you have a version number of the firmware? When you press ESC key you can see the version from the screen.

unkyulee, 2025-03-03:
I have update the release and added the fix. I do not know if the fix will resolve this shift key issue. But, worth trying. 

https://github.com/unkyulee/micro-journal/releases

Please, try downloading the rev.5 firmware and place it inside the sd card the reboot the device. 

I didn't have chance to test nor see the root cause of this issue. It's just a guess where could have been the impact. I really hope this could solve the issue. If you are willing to ship me the keyboard for me to test it, I can look into the cause and fix it. Otherwise, I am not able to test the cause at the moment.

Ryonne, 2025-03-03:
It worked! Thank you so much. You've really gone above and beyond with this product.