unkyulee/micro-journal #48: Boot Loop (Rev 2)

corycaean, 2025-03-04:
I have a new Rev 2 whose functionality I love, but I've not been able to get it to boot unplugged since I got it. At first I thought it was a bad battery, but I got a good one and it's still not booting. Once it boots, I can unplug it and it crashes when I try to open any file. I know the battery life isn't intended to be long, but I wasn't planning to use it plugged in forever. Any advice/a fix? See [here](https://streamable.com/l1dyhs) for a look at what's going on. (The video expires in two days, so sorry to the future people).

unkyulee, 2025-03-04:
If the device shutsdown during the booting process. It means the battery is
not able to push the current that is required during the boot process.
Booting. If the battery is not the issue then it could be the power supply
that may not be able to produce sufficient current.

But when it is plugged to external power then it boots? Then the power
supply may not be the cause. Unfortunately, i think you may need to try
with another battery.
Un Kyu Lee

corycaean, 2025-03-04:
Thanks for the reply. What you're saying certainly makes sense. See any red flags with [this battery](https://www.18650batterystore.com/products/eve-35v-18650-3500mah-10a-battery)? It's the one that's currently not working. Want to know how I ordered wrong.

unkyulee, 2025-03-04:
They look ok. When I saw 3500 mah, made me feel a little bit suspicious but, I guess it's in the range. 
But it could be that the battery may not be fully charged or can't be fully charged. By any chance you can measure the voltage of the battery and see if it is in the range of 3.5 ~ 4.2 volts?

If the issue continues and can't find a solution. You can ship it to me and I can have a look. To get the shipping address, send me a message in tindie.

corycaean, 2025-03-04:
I don't have a voltmeter unfortunately, so I may take you up on that offer to have a look. I have another battery coming today, so I'll try that first. (It'll be my fourth attempt though, so perhaps I'm singularly bad at buying batteries!)

unkyulee, 2025-03-04:
Make sure to charge the battery. LiPo batteries doesn't come fully charged for safety reasons.

corycaean, 2025-03-06:
So the new (fourth) battery offered no better results, but before shipping it all the way back to you I thought I'd just muck around with it and see if I could figure anything out and here's my current read. 

I opened up the case and plugged micro-usb cables into the Pi and display controller and then into my anker USB battery pack, and it booted like a champ. The problem doesn't seem to be with either of those components. My 18650 battery also seems to be holding a charge (at least according to the indicator). So my current theory is that the power supply just isn't pushing enough juice from the battery to the Pi/display during the boot process. That seems to suggest either a faulty wire or a faulty power suppy to me, and those are both problems I feel like I can probably fix. I don't have experience, but looking into soldering, I think I could swing it. Found the part on Aliexpress and ordered it, but I think it needs a week to arrive. 

I ordered a multimeter from Amazon, and I think it should be here tomorrow. I can confirm then. But do those seem like the most likely culprits to you? Thanks.

unkyulee, 2025-03-06:
I would check vout of the power supply and see what is the voltage during boot. It maybe going below 5v that could cause the reset. There should be a screw on the supply that can adjust the voltage. Very gently turn by 1mm and check again the voltage to see if it increases. Don't push it beyond 5.8v because that can burn the display. Probably best to remove the display before changing the voltage.

corycaean, 2025-03-06:
That would make a ton of sense as the culprit; that thing looked suspiciously twistable. I'll give that a shot when I get home from work today. Thanks again for helping a total novice.

corycaean, 2025-03-06:
Alright, update: this seems to have fixed it. 

At 5.3v: bootloop
At 5.4v: boots but crashes to bootloop a few seconds later
At 5.7v: boots and stays booted
At 6v (I accidentally turned it too far): Pi boots but display goes non-responsive

Luckily when I went back down from 6v to 5.7v, it started working again. Thanks again for your help!

unkyulee, 2025-03-07:
What a relief for me. I feel like we have solved a difficult puzzle and happily lived ever after.