unkyulee/micro-journal #49: Keyboard stop working

violy, 2025-03-13:
Hello there !

The keyboard (and the wheels) just have stop working, after I'm trying to try to follow the [Hook’s Key Map Guide for the Macro Journal Rev 2 Using Vial](https://github.com/unkyulee/micro-journal/blob/main/micro-journal-rev-2-revamp/keyboard/Hooks-Keymap-Guide.pdf)

I've probably made some mistake… one of the biggest mistake is probably that I've **Power ON** the MJ, when it was connected to my MacBook Pro… 😭 (I've started Power OFF, but nothing appears on Vial)

Actual state : 
- MJ rPi is booting. 
- screen is displaying the `ranger` home. 
- I've an SSH access working over wifi. 

I've try some commands over ssh … 
```
microjournal@microjournal:~/microjournal $ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

So, how can I make a better diagnostic, and troubleshoots ?

Is it a physical damage ? On which part ? The Raspberry Pi Pico microcontroller ? The Keyboard PCB ? 
What is my best option ? Happy to learn !

Thanks. 
Arthur.

unkyulee, 2025-03-13:
You may need to replace the keyboard micro controller. Here is the guide about DIY builds. 

https://github.com/unkyulee/micro-journal/blob/main/micro-journal-rev-2-revamp/build-guide.md

Have a look at the keyboard control assembly to get more details.

violy, 2025-03-13:
Ok, thanks for your very quick answer ! I think I can do it ! 

If I resume the steps : 
- buying a [raspberry pi PICO](https://www.raspberrypi.com/products/raspberry-pi-pico/) controller.
- unsold all wires on pins of the damaged controller (keyboards, two wheels and USB).
- sold all existing wires on pins of the new controller.
- flash the QMK-Vial Firmare on the Keyboard controller. (but usb wiring seem's not well working on MacOS… maybe can I use another RaspberryPi and use Vial on Linux ?)

Is that right ?

unkyulee, 2025-03-13:
Wow, that's very precise description of what needs to be done to repair!

violy, 2025-03-20:
Hi there, some updates 

- I've order and receive the Pico controller. 
- I've unscrew the keyboard to find the existing controller.
- I've discover that the controller is glued, and every controller pin soldered is also glued.
- I've remove the glue and detach the controller. But I cannot unglue / unsold the controller's pins…
- From here, I'm searching the best method to connect the new controller… not so easy !

[image]
[image]

The method I imagine to continue :
- **Mark every wires with my Posca**  (paint markers) to **color code each pin**. 
- Cut every wires as close as possible of the glue, avoiding the maximum loss. 
- Strip the wires to sold them to the new controller.
- Sold existing (and re-stripped) wires to the new controller. 

What do you think about it ?

unkyulee, 2025-03-20:
You can gently pull it towards the outside and the cable will extend till the right hand side where you can settle it on the desk or at the edge of the enclosure. 

For the glue... Hmm... What I do is just stick the soldering iron in and it does melt the solder when made contact. Then desolder one wire then solder to new pico. Repeat until all wires are transferred. 

Glue may smell a bit nasty while melting, so I wear masks. But those are hobby glues so shouldn't be toxic. Just better to be safe.

This is what I do when repairing similar cases. Then once all the wire is transferred, i make sure to clean solder tip and tin it with fresh solder to protect the tip from rusts.

violy, 2025-03-21:
Thanks again @unkyulee !

Before do that, I've also try a _last chance action_ : **Plug the existing Pico controller into USB of a <ins>Windows PC</ins>** …

I connected the Pico controller via USB to my Windows PC (where Vial GUI was now installed) and, surprise, the MicroJournal screen started up (the switcher is OFF!). I tried the keyboard keys and the scroll wheel, and noticed that it worked – not on the MicroJournal ! – but on my PC! Vial recognized the keyboard! After that, I unplugged the USB cable, and the MicroJournal keyboard worked perfectly!

It's now working again! 😅

My conclusions:
- Be careful when you connect the Pico controller USB!
- Prefer using a Windows PC for Vial rather than a Mac.
- I've fail to install Vial GUI on Raspbian (rPI B4).