unkyulee/micro-journal #50: [DIY BUILD] - Rev2: Revamp - Display seems to draw too much current.

markmcgookin, 2025-03-18:
Hi, 

I was wondering if you have come across this issue before. I have just finished my DIY build of the rev2:revamp, and was doing some final tests before putting the case all together.

Everything has been working fine, but when I plug in both the display and the pi together the power supply board and the battery both get very hot (hot enough to melt the hot glue holding them in place in under a minute), and nothing boots, I can hear a feint whine when I turn things on.

If I only plug in the pi it works fine and I can SSH into it, and I can plug in and power the display separately and I can boot into the OS and use away.. but I cannot seem to run both at once. I ordered the parts listed on the build guide from AliExpress, I checked to make sure I was getting 5v out of the power supply board (5.4 exactly) - I have tried two 18650 cells, one from AliExpress (2000MAh) and one more 'premium' one (2800MAh) both have the same result. 

Have you ever encountered this issue? I don't want to close up the case until I am sure it will be safe to use.

markmcgookin, 2025-03-18:
Ok, I should have tested this before posting... I _think_ that the issue was one of the usb leads going from the battery if I connected it to the pi or to the display neither booted. If I used the other lead, each thing would boot (a blue light shows on the display... so I think I need to get and prepare another micro usb lead, and will replace the battery charging module and switch as well just to be safe.

unkyulee, 2025-03-18:
I hope I am not too late or maybe what I will say would cause a bit of sadness. 
Currently I am going through a process of removing the battery from rev.2. 
Reason being that the power supply is not a good fit and also just one battery is not good enough to support raspberry pi. Also, causing numerous issues from the users while installing and using it. I feel like it's actually worse to have an internal battery in rev.2. 

Feels like delegating entirely to external portable battery seems to make more sense. I would replace the power supply to a simple usb c female port and delegate the power supply entirely out side.

markmcgookin, 2025-03-18:
Right... I'll close this now. I rebuilt the power module with new components and everything is working as expected!

closed 2025-03-18 by markmcgookin