unkyulee/micro-journal #51: Rev 2 - keyboard doesn't consistently work

bravethefog, 2025-03-19:
Hi!

I just got my Rev 2 yesterday, and I'm super happy with it! Thank you so much for the beautiful work.

As of today, the keyboard and the dials aren't consistent with working, on battery and plugged in, both back and side USB-C ports.

I thought it was at first tied with how I was shutting it down, either through shutdown.sh or the switch. But then I did the keyboard test of plugging it in through the side USB-C to my PC and it picked up the keyboard just fine. After I did this and unplugged, the keyboard and dials started working again.

However after another reboot, it stopped working again.

Is there something I'm missing in my troubleshoot?

Thank you!

bravethefog, 2025-03-20:
Hi again - just wanted to update.

I let it charge for a few hours to attempt the keyboard-PC method through the side USB again, but this time upon plugging in, Windows gave me a popup saying it doesn't recognize the device. I don't know if that helps in anyway.

unkyulee, 2025-03-20:
I feel like there are more complicated issues going on. 
I have a feeling that the battery is completely out but probably best to check out in detail.
Would you send me a message via tindie message, or find my email address from my youtube channel and send me a private message please? So, I can give you the shipping address

https://www.youtube.com/@unkyulee

bravethefog, 2025-04-01:
Hi!

I just wanted to update that after trying different battery brands, everything seems to work now? I'm going to continue using it for the week as I'm still suspicious of this. So I'll keep this ticket open just in case, and I already have your address so I'll let you know on here/Tindie if this becomes a false positive.

P.S. I got the Rev 7 last week and I'm in love. Thank you! <3

unkyulee, 2025-04-02:
You can keep the ticket open, many others with similar issue can find this story useful

My fingers are crossed