unkyulee/micro-journal #53: Troubleshooting Keyboard disconnected

violy, 2025-03-31:
Hello, 

As mention in my previous issue #49 I've **not** replace the Pico controller, because the keyboard finally works again after I've plug the Pico controller to my PC via USB. After that, I retry to reconnect also with my _old_ MacBook Pro (gen. 2013) and Vial also works now on MacOS !
I've configure some news Keys, with an French Mac layout variant, and it's worked fine ! 

But… **randomly**, the **keyboard** **disconnect**. 
So I plug again the USB wire to my Mac, with Vial, and unplug. And the keyboard is working after that…
Problem solved… not really, because the duration of the connection cut off very quickly … from one to 5 minutes. And sometimes, without any interactions. 

I've try — as mentioned here #51 — **on battery** and **power plugged in**, the keyboard is disconnecting. 

As the ssh connection works after the keyboard disconnect, I try to get the system message log `dmesg`. 
Bellow is the details with timestamps : 

```log
[  174.248689] usb 1-1: USB disconnect, device number 2
[  174.780629] Indeed it is in host mode hprt0 = 00021501
[  175.116536] Indeed it is in host mode hprt0 = 00021501
[  175.384532] Indeed it is in host mode hprt0 = 00021501
[  175.652528] Indeed it is in host mode hprt0 = 00021501
[  175.980504] usb 1-1: new full-speed USB device number 3 using dwc_otg
[  175.980760] Indeed it is in host mode hprt0 = 00021501
[  176.108558] Indeed it is in host mode hprt0 = 00021501
[  176.376536] Indeed it is in host mode hprt0 = 00021501
[  176.725680] usb 1-1: New USB device found, idVendor=feed, idProduct=0000, bcdDevice= 1.00
[  176.725696] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  176.725705] usb 1-1: Product: unkyulee/ortho_48
[  176.725713] usb 1-1: Manufacturer: Un Kyu Lee
[  176.725720] usb 1-1: SerialNumber: vial:f64c2b3c
[  176.726153] ERROR::dwc_otg_hcd_urb_enqueue:502: Not connected
[  176.726407] usb 1-1: USB disconnect, device number 3
[  176.904538] Indeed it is in host mode hprt0 = 00021501
[  177.092506] usb 1-1: new full-speed USB device number 4 using dwc_otg
[  177.092724] Indeed it is in host mode hprt0 = 00021501
[  177.220541] Indeed it is in host mode hprt0 = 00021501
[  177.488533] Indeed it is in host mode hprt0 = 00021501
[  177.837685] usb 1-1: New USB device found, idVendor=feed, idProduct=0000, bcdDevice= 1.00
[  177.837702] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  177.837711] usb 1-1: Product: unkyulee/ortho_48
[  177.837719] usb 1-1: Manufacturer: Un Kyu Lee
[  177.837726] usb 1-1: SerialNumber: vial:f64c2b3c
[  177.840621] input: Un Kyu Lee unkyulee/ortho_48 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1:1.0/0003:FEED:0000.0004/input/input7
[  177.964871] hid-generic 0003:FEED:0000.0004: input,hidraw0: USB HID v1.11 Keyboard [Un Kyu Lee unkyulee/ortho_48] on usb-3f980000.usb-1/input0
[  177.965288] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 0, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965338] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 7, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965377] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 6, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965451] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 1, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965488] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 2, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965523] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 3, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965623] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 5, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965657] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 0, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965692] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 7, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965745] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 6, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965778] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 1, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965813] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 2, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965865] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 3, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965897] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 5, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965932] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 0, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.965983] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 7, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.966016] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 6, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.966050] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 1, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.966101] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 2, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.966133] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 3, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.966168] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 5, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.966199] usbhid 1-1:1.1: can't add hid device: -71
[  177.966223] usbhid: probe of 1-1:1.1 failed with error -71
[  177.966511] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 0, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.966545] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 7, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.966582] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 6, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  177.966640] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 1, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[... 67 lines cut ...]
[  180.255432] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 0, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  180.255487] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 7, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  180.255538] usbhid 1-1:1.2: can't add hid device: -71
[  180.255576] usbhid: probe of 1-1:1.2 failed with error -71
[  180.256011] usb 1-1: USB disconnect, device number 6
[  180.524577] Indeed it is in host mode hprt0 = 00021501
[  180.712509] usb 1-1: new full-speed USB device number 7 using dwc_otg
[  180.712768] Indeed it is in host mode hprt0 = 00021501
[... cut ...]
[  182.168546] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 0, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  182.168591] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 7, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  182.168637] ERROR::handle_hc_chhltd_intr_dma:2212: handle_hc_chhltd_intr_dma: Channel 4, DMA Mode -- ChHltd set, but reason for halting is unknown, hcint 0x00000002, intsts 0x06000021
[  182.376499] usb 1-1: device not accepting address 8, error -71
[  182.376555] usb usb1-port1: attempt power cycle
```

unkyulee, 2025-03-31:
This issue is most likely related to the connection of the usb from pico to usb cable is not in a great state. Most probably solution i had success with is to resolder the 4 wires to the pico board. 

If the problem persists i can make a repair. Please contact me via tindie so that i can send you my contact privately 

Un Kyu Lee

violy, 2025-04-01:
Thank you. 

So before contact you via Tindie, I will try again, and seeing if the problem persists. 

Arthur.

violy, 2025-04-15:
After many days without disconnecting, I close the issue. 

I've put a painting tape over the USB port to avoid any electrical perturbation …
[image]

Thanks.

closed 2025-04-15 by violy