unkyulee/micro-journal #55: Freeze at launch

Jlnnnn, 2025-04-18:
Hello,

So I just received the Rev 7 yesterday morning and seem to experience some kind of freeze at launch.

Here is what I did or observed so far: 
- I firstly followed the guidelines to set it up, inserting my LIPO battery (this one: https://www.akkuteile.de/en/keeppower-r-18650-3120mah-3-6v-3-7v-li-ion-battery-protected-positive-terminal-elevated_12034_2168) 
- Proceeded to charge the Rev 7 with a USB A to C cable ; red dot was on. 
- At 75% charging, I started the Rev 7 for the first time ; the screen flashed, and I could type a sentence, but then the first freeze happened - the underscore didn't flash anymore, I couldn't write, clear the screen or go to sleep mod. I turned it off. 
- Left the device charging for 4 hours. 
- Pulled the SD Card to prepare for the Google Drive sync, inserted the SD Card back in the Rev 7. 
- I started the Rev 7 for the second time. Went on the menu to set up the Wifi, and on the Wifi page, the freeze happened. I turned it off and on. Each time I was turning it on, the freeze was happening faster, and ultimately within a second. I turned it off for the night. 
- This morning, I started it again. Weirdly, the freeze happened after maybe 30 seconds so I could type a few sentences. After re-starting it, the freeze is again happening within a second. 
- I pulled the SD Card and reverted the json file to its og content, inserted the SD Card back in Rev 7, but it didn't change anything.

unkyulee, 2025-04-18:
I am sorry that you are experiencing these errors. The cause of this is due to the USB cable that connects the keyboard controller and the display. I have soldered the cable by hand and it seems like it's creating electrical interference and results into a disconnect. Quick solution is to open it up and remove the glue that is on the USB connector on the display. Once the quick solution is applied see if the problem is resolved. If not, then other option is to replace to the premade cable that I have linked below.

Premade cable link:
90 degree angle male connect to male. 
Male to Male Up, 0.3m
https://it.aliexpress.com/item/1005004357769641.html

To open up the display, you will need to remove the right hand side compartment

1) Remove the top two screws on the right side
2) Remove the bottom two screws on the right side
3) Remove the 2 columns of keys and remove the two screws on the keyboard PCB
4) then the right compartment will become lose and gently pull it out. Be careful with the cables
5) Remove the display enclosure. Remove the rubber rings
6) Remove the 4 screws on the display enclosure. 
7) Open it up and you will see a USB cable. Check if this connector has a glue on it then remove the glue very gently with your nails. 

I am very sorry for your experience. It's still a bit early release and it's not long that this issue has been discovered.

Jlnnnn, 2025-04-22:
Thanks for your quick answer and sorry for my late reply, my schedule is busy lately and I struggle to find time.
I opened it up today and found at least three areas where I can see glue. Being unsure which one leads to the display, could you indicate the right one to me ? Here is a picture: 
[image]
Thanks in advance.

unkyulee, 2025-04-22:
What you are looking at is battery enclosure. The glue that need to be removed is the one inside the display enclosure. 
You need to get into the display as the picture. Make sure to remove the rubber ring before opening like that. 

At the moment, I don't have the build at my hand... so it's difficult to make a precise picture. 
To get access to the display enclosure. You will need to remove several screws on the right-hand side compartment.

two screws on the top
two screws on the bottom
two screws on the PCB

[image]

Once all the six screws are removed gently pull out the right hand compartment
You will be able to remove the display enclosure from that point. 
Make sure to pull out the rubber ring before
and then remove 4 screws on the back of the display panel
You should see as below picture (Make sure to remove the rubber ring)

[image]

Then clean up the glue on the connector you see with your nail, very gently. 
Before putting things back, turn on the device and see if the keyboard activity is restored.

Jlnnnn, 2025-04-23:
Thanks for the insight. I'm keen to try fixing it myself, proceeding step by step and cautiously. 

As things stand, I have removed screws 1, 2, 3, and 4, but screws 5 and 6 are out of sight. Shall I remove the keyboards keys to access them? If so, iss there a "good way" to remove the keys?

unkyulee, 2025-04-23:
There should be white clip shaped thingy inside the keycaps bag. That will give you a grip to the keycaps and you can use it to pull them out.

Jlnnnn, 2025-04-23:
[image]

It seems to be missing unfortunately. I guess a standard key caps remover from amazon would do the trick?

Jlnnnn, 2025-04-23:
Followed each step until I found and removed the glue, a bit chunk indeed:

[image]

[image]

I then light up the Rev to try it, but had just enough time to type "Fixed" before it froze again, as it used to before :-(

unkyulee, 2025-04-23:
Hmm..I am out of options... i think only way is to make a repair on my end. I will contact you in private message to follow up with you. Sorry for all the hassles.