unkyulee/micro-journal #57: Can't go back to edit earlier text

Flybell, 2025-04-23:
Dear Mr. Lee,

Could you please explain the concept of screen buffer to me? After syncing my file, I could not go back to edit the text I wrote in the beginning. My text file has 2664 characters, I can see it in google drive, but I can no longer access most of the text in the beginning. In another issue, you explained that this has something to do with screen buffers. But I do need to go back and forth to read what I wrote as I brainstorm an article. Is it not possible to do that once a text exceeds 2000 characters?

unkyulee, 2025-04-23:
Micro journal was initially designed to focus on writing forward and didn't have any edit features except backspace. That's the reason why there is a screen buffer which will just have what is on the screen. File saved in the SD card, will have the full text, of course. But what will be shown on the screen is limited for this reason.

Later navigation feature was introduced which allows to go back and forth and edit texts in the middle. But due to the initial design constrains, it is implemented within the screen buffer which will not have ways to navigate beyond its buffer limit. 

Screen buffer size at the moment is 4000 characters. When you reboot the device, buffer will be filled up to 2000, so that there is a half full of text to write further and half full that can be looked beyond. 

I have plans to create a software that can be called a proper word processor. Aiming to be in the level of what Pomera series provides. It is going to be a big project in itself to achieve, so I can't promise you that will be delivered in a short period of time. I personally am eager to get the software to improved up to that level.

Flybell, 2025-04-23:
Is there a way for us to tinker with the system and change the screen buffer?