unkyulee/micro-journal #58: Issues Charging and Syncing to WiFi

stefanesa, 2025-05-04:
Just received my Rev 7 and love it!  Despite two minor issues.

(1) When I got it my battery wouldn't charge, despite having it plugged in for several hours.  I had to take it out and charge it through other means.  The indicator for the charger doesn't change from red.

(2) When I first set it up I was able to connect to a wifi, but since then when I try to go to the wifi menu it just goes back to the last document I was editing.

unkyulee, 2025-05-04:
For the wifi configuration, i believe it is a software bug, i will have it fixed soon. 

For the battery not charging, i suspect it could be related to the type of the battery. Since it is supplying the power to the device, it should be able to charge it. 

Would you attach a picture of the battery please?

unkyulee, 2025-05-05:
I have made an update to be able to wipe out wifi configuration so that the feature can be restored. 

https://github.com/unkyulee/micro-journal/releases/tag/1.0.18

Please, download the "firmware_rev_7.bin" file and save it to the SD card and the restart the device to have the firmware updated.

stefanesa, 2025-05-05:
Thanks, I'll update the firmware ASAP.

Here's the battery I'm using: https://www.18650batterystore.com/products/panasonic-ncr18650ga-ga6?srsltid=AfmBOorCsRaz2YIqQVRrhB8lg2w2pvIsTfN544t5I5qF8t35fX-20jt4

unkyulee, 2025-05-05:
Hmm.. the battery seems fine tho. If the red light is coming out then it is a good sign. It is very slow charger so, it can be that it is taking long to charge fully.

stefanesa, 2025-05-10:
I'm still having the same trouble with syncing - I just updated the firmware and when I hit [w] it just flashs and goes back to my document

unkyulee, 2025-05-10:
There should be a menu item called reset, which will wipe the wifi
configuration.
Un Kyu Lee

stefanesa, 2025-05-10:
I don't see that menu item -- I can't actually access the wifi menu

Here's a video: https://youtu.be/oN6xY_jmLZY

unkyulee, 2025-05-10:
Seems like the firmware isn't the version that I was talking about. When
you updated the firmware did you see a message on the screen about the
firmware update?

New firmware should show you a reset function which will wipe the wifi
configuration which is causing the crash when entering the wifi menu.
Un Kyu Lee

stefanesa, 2025-05-10:
Well, it looks like when I copied the new firmware over a (1) got added to it's name, so it didn't sync.

Thanks for the help!

stefanesa, 2025-11-21:
Hey - here are the two chargers I used.  At this point the lights don't even turn on while charging.  The apple charger is the 70w

Is there a maximum wattage I should use to charge?  Do you think replacing the battery will fix it, or is there a chance I need to replace the power board?  Can you send a link to where to buy a replacement if so?

[image]
[image]