unkyulee/micro-journal #59: Sync to Wifi menu option disappeared (1.0.18.r4 REV.6)

zeinrich, 2025-05-05:
I've had my REV 6 for a month or two and have been using it every day.

I recently updated my REV.6 to 1.0.18.r4. I successfully synchronized a file today (maybe the first file on the new firmware?), and now the menu option to sync over wifi is missing. I also noticed that my preferences were reset: I usually have green text, and it was reset back to gray. I checked my wifi settings, and they are still there and still correct.

Any advice on how to get the menu option back?

zeinrich, 2025-05-06:
Nevermind! The reset of my preferences should have clued me in. It lost the URL to sync with. I have added that back in and now it syncs.

There does seem to be a bug that resets preferences, maybe during a sync? Not sure.

unkyulee, 2025-05-06:
Sounds like config.json has been deleted from the sd card and also seems like wifi configuration has been deleted. Feel like a big reset has happened somehow. I may not be able to pin point to exact cause.

zeinrich, 2025-05-06:
After playing with it a little more, I can say with some confidence that syncing sometimes resets config.json.

Is it safe to rollback to an earlier firmware?

unkyulee, 2025-05-06:
it is safe to rollback to earlier firmware

zeinrich, 2025-05-19:
Additional observations, if they help at all:

It appears that the reset is happening during startup:

[3456] Creating default config.json template
[3490] Config updated successfully.
[3521] Total heap: 288932
[3530] Free heap: 253680
[3540] Total PSRAM: 8386247
[3549] Free PSRAM: 8359051
[3559] ILI9341 Display SETUP
[3568] DISPLAY ILI9341 SETUP

I haven't seen it happen at other times.

I don't see 1.0.18.r3 REV.6 available, and I'm afraid I didn't back it up. How might I download an earlier 1.0.18 release?

unkyulee, 2025-05-19:
You should be able to find past released in the following link:

https://github.com/unkyulee/micro-journal/releases

according to the log it fails to find the config.json file in the SD card and it went into creating a new one. This could be the display board itself (which contains the SD card reader) or SD card is not stable. As I see that the file is written correctly after... so could be SD card has some partial failures. Easier way is to replace the SD card... or when before removing the SD card from the PC you might want to press "Eject Media" this way it will properly close the file system and ... maybe it might recover some misalignments. 

I would try to do Eject Media to make a proper closure on the SD card and try again.
If it continues, I would try to format the SD card and try again
... if it continues still, then I would try to replace the SD card
if it continues... then I would suspect that the display module may have a problem. 

If you agree to ship it back to me, I can have a look and replace the necessary components

zeinrich, 2025-05-19:
I will try all you suggest!  (I have been practicing good card ejection hygiene, so we can check that box.)

Thank you so much! If I've exhausted all of these options, I will reach out again.