unkyulee/micro-journal #60: Micro Journal rev.2 loading up then freezing (or keyboard not working)

jd-eck, 2025-05-29:
Hi there,

I received my Micro Journal rev 2 yesterday, and have been trying to use it, but not getting anywhere. The system boots up, goes through normal linux startup lines, then loads the dashboard. But I can't do anything from the dashboard, I'm not getting any response from the keys, and I've tried any and all options possible, and nothing happens. I have tried the knobs as well, though I'm not sure they would be expected to work here, but still nothing happens.

I've tried:

- Using 3 different cables
-  Using a different power sources, an old usb-a charger, a fully charged usb power-bank with a plain usb-a output, and a different usb-a non-pd charger.
- I tried removing the SD card and re-inserting to be sure it is well seated.
- I tried the power down command, and nothing happened there either

Other than 1 cable, most combinations of the above would load the system to the dashboard, and then nothing else was possible.

I'm a little hesitant to flash the microSD and work through the process of configuring it, but if that's the next best step, I do think I'm capable of doing this. 

Please let me know what else I can try.

jd-eck, 2025-05-29:
Thank you for the quick reply! I was able to open it, and I see the cable and connector is micro-usb. The cable was connected securely, so maybe the cable is the issue. That link opens to a product not found page (maybe not available in US?), so would it be something [like this](https://www.amazon.com/Android-Connector-Platinum-Controller-Accessories/dp/B07TYQLQQ2?sr=8-12)? Micro-usb male to micro-usb male data cable?

unkyulee, 2025-05-29:
The link you send, should be correct one... but I am afraid that the angled connector would probably will interfere with other cables. 
Try this type of cable below

https://www.amazon.com/CableCreation-Micro-Compatible-Android-Tablet