unkyulee/micro-journal #61: No way to build esp32 with custom 30 keys keyboard firmware

kolyan94, 2025-06-22:
Hey! So I made a diy micro journal with the custom keyboard but can’t find anything in the platformio file. Looks like this version was deleted :(

unkyulee, 2025-06-23:
I have to make some assumptions givens your description. You have made a split keyboard matrix that is intended to create rev.4, correct? and you are looking for a solution for the firmware and you are not finding it? Or you found it but it's not with esp32 s3 board? 

If I were you, I would look into rev.6 code and see if you can modify the matrix to fit your implementation. 

https://github.com/unkyulee/micro-journal/tree/main/micro-journal-rev-4-esp32

kolyan94, 2025-06-23:
Correct, I tried to use my old esp32 (not S3), but it was hard to build. But I already ordered S3, so will try soon one more time! Thanks

closed 2025-06-23 by kolyan94