unkyulee/micro-journal #64: Syncing stopped working (Rev.7)

Gabey8, 2025-07-26:
I am updated to the most current Rev.7 firmware, as of July 26.

I can see and use the Bluetooth keyboard option (and am using it now).

However, syncing has stopped working. The first thing I did was re-enter my wifi info, just in case that had gotten corrupted somehow. 

Then I created a new Google Sync app deployment, just in case something was wrong with that. 

There is still no file appearing in my /uJournal folder when I hit S to sync.

It even LOOKS like it's working. It tells me it's connected to the wifi, and says it's sending the file. But no file ever arrives.

This happened one other time, a few weeks ago. On 7/16, I updated the firmware to what was the current version for that day, and syncing resumed working.

I am wondering if the new firmware has a bug in it where syncing is concerned.

unkyulee, 2025-07-26:
I just tested with the new firmware and seems to be working. 

I have a feeling that sync url may not correctly set or the sync script is not installed correctly. 

Micro Journal does not know if the file is received well from the sync script or not. It will say sync is completed when data is sent through internet. So, if you have seen the sync complete message. Most likely means that micro journal has a good connection to the internet. 

Probably, sync appscript hasn't received it or didn't reach the appscript. One way to make sure is to open the appscript and look for "Execution" and it will show you the history of access from the micro journal. If you don't see any transactions there it probably means that you have not used the correct sync url. Or the deployment is not set correctly.

You can check in the manage deployement and copy Web App url, and update the sync url in the config.json file. Also make sure that the appscript is configured to Excute as "me" and Who has access "anyone". If all this is too cryptic for you, delete the appscript and try to follow the guide step by step again. Usually (even for me) first try never works. Second time is a charm, or even third.

Gabey8, 2025-07-28:
I got it to sync by taking the config.json file from the Rev.6, which I had just verified was syncing properly, and editiing it to remove some items that the Rev.7 doesn't use (startup animation and one other item). 

I have a small collection of web apps in my Google Drive now, lol, but as long as at least one of them is working, I'm happy. :)

closed 2025-07-28 by Gabey8