unkyulee/micro-journal #67: Help, Alt key functions don't work

luckylita, 2025-08-16:
Hello,
I have recently update my Micro Journal 6 and the new keyboard layout sadly did cause confusion and frustration for me when I turned it on. I didn't want to go through pulling all of the keys and rearranging them again since I got used used to its original keyboard layout from the day I got it back in June. I did create a keyboard JSON file with the keyboard layout that I was using. Everything worked except for the Alt key functions. When I hold the Alt key and try pressing for example the "q" key, my MJ 6 types out the letter "a" instead... I'm stuck with a jumbled Alt key layout that doesn't match what I'm typing and I have no idea how to fix it... ;  ;
[image]
[image]

[image]

unkyulee, 2025-08-26:
There has been breaking changes in the latest firmware. Json format has gone through some changes. 

Here are the details.
https://github.com/unkyulee/micro-journal/blob/main/micro-journal-rev-6-one-piece/keymap.md

In essence, you may want to change the "alt" to "lower", and "alt-shift" to "lower-shift"

luckylita, 2025-08-26:
Thank you! The keyboard layout is finally fully mapped to the JSON settings I modified. I was confused on why "alt" doesn't work anymore, and I just tested the functions after I changed it to "lower".

closed 2025-08-26 by luckylita