unkyulee/micro-journal #68: Rev 6 issue on DEV 1.0.19 release, screen refresh issues, all written text appears on bottom line

tallzilla, 2025-08-19:
Hi there!

I just tried 1.0.19, firmware_rev_6 bin on a newly-assembled Rev 6. I have no issues with the stable rev 6 software.

With 1.0.19, when I boot up on a fresh install, the device appears to have issues refreshing the screen.

Attaching a video illustrating what is happening. As I would describe it:

The screen never fully wipes so:

1) Going from the opening winking face gif to the writing interface, the winking face gif is still present and wasn't cleared
2) only the bottom line of written text is visible, and writes over the last line

tallzilla, 2025-08-19:
Having trouble uploading a MOV through github, here's a shared google drive video:
https://drive.google.com/file/d/1rMZEAvFx8zivnB6oMZ-Yl8uhMZxAj-zT/view?usp=sharing

unkyulee, 2025-08-26:
Hmm.. this is certainly weird. Given your description that it is caused by the software changes. But, I can't really reproduce the issue in my devices... This is sad that is happening. I should be having some updates on rev.6 in couple of weeks and new firmware should be released by then. 

As I can't really pin point to what is causing this issue... I can't assure you that fix can be found... 

I would recommend to use the stable version for now and please check out the new firmware in couple of weeks.