unkyulee/micro-journal #71: Rev.4 saving glitch: it only saved what's in the buffer

Gabey8, 2025-09-13:
OK, the good news: I now have the current Rev.4 firmware. I just updated it this afternoon.

The other news: the device seems to only be saving what's in the buffer when I create a file.

I take the device name literally, and use them for journaling. :) I created a rather verbose entry this afternoon (What, me verbose? I'm sure everyone is shocked, lol). There's a lot going on, and as all who journal can attest, it's a good place to have a proper rant while processing one's thoughts.

When I went into Drive Mode, to copy the 0.txt file into my actual journal, I discovered that the only text in the 0.txt file is what's currently in the buffer. The whole beginning of what I wrote has vanished. While this could be the universe's way of telling me not to rant, I'm more inclined to believe that the universe is just telling me to report a software glitch.

HookPro1, 2025-09-14:
I just ran a test on this and something very odd seems to be going on. I typed in text into a blank file and periodically would check file info for word and byte count and then test the buffer using page up.  Everything was in the buffer until after I hit 349 words and 4 852 bytes. A little more typing and suddenly the word count became 178, the buffer was missing all but those (I presume) 178 words, but the byte count was 4852 (so it kind of knows the content missing from the buffer was there). Sure enough, when I checked, the file only had the 178 words in the buffer. Also, weird, Send didn't work,. Send works fine on other files that didn't exceed their buffers. Very strange.  It almost sounds like when the buffer is exceeded, The Rev 4 not only trims the older text in the buffer but in the file as well.  And, whatever happens, Send is convinced there is nothing there to send.  I'm hoping maybe this data will help.

unkyulee, 2025-09-14:
I have updated the code and it seems to be resolved. 
https://github.com/unkyulee/micro-journal/releases

I am very sorry this issue and thank you very much for reporting the issue with great detail in how you got the error. Thanks to your explanation, I was able to reproduce the bug, and found the root cause and fixed it. 

Problem was that while the saving is on going, it was trying to load the file at the same time. This operation was causing a critical conflict on the file system and made to reset the content.......... I had this safety system in place but in the new version which went through entire rehaul... forgot to bring that safety code back in.....

Safety code is back in and it should be stable than before. Please, try it again, and let me know if the issue is fixed.

Thanks for your patience

Un Kyu Lee

Gabey8, 2025-09-14:
How long should a FW update take?

I want to figure out if I should keep waiting or if the update is hanging.

I went into the FW update part of the menu, copied the latest FW onto the drive, and the Rev.4 screen has been dark for at least five minutes. 

Where can I find the FW from a couple updates ago? I'll see if I can grab that and restore writerdeck functions until I can figure out why this update doesn't seem to want to finish.

Gabey8, 2025-09-14:
It's totally unresponsive with a dark screen (not off, because if you shield it from other sources of light, there's still that tiny bit of light that emits from a screen that is showing blackness, but is not off). None of the buttons or keys will wake it back up.

Unplugging it, then turning it off/on, gets it back to life, but it doesn't appear to have updated the firmware.

Gabey8, 2025-09-15:
OK, the good news: it is definitely changing the filespace. So the ALT key works.

But I still have my oversized file from yesterday, the one in 0.txt that started losing characters from the top of the file as it exceeded the buffer size. And, upon further experimentation, adding to that file causes the text at the beginning of the file to disappear as it "falls out" from the buffer. So the current FW update didn't take.

HookPro1, 2025-09-15:
If Alt key works it could mean you don't have new FW.  Lol. The only way to know for sure is to do the update again. I had no trouble installing the firmware, but the first time, as `i said, it wiped my gifs so I thought the Rev 4 was dead not because it was but because I had a black screen with no animations.  

I would not use the same filespace. That text is gone and when I ran my test, once the file lost the text, it wouldn't Send either. And it still showed the correct bites, but I don't think that means the data was retrievable. I don't think that fiilespace makes a good test of the new firmware. 

I'm about to run another exceed the buffer test and will see what happens. It should behave this time.

HookPro1, 2025-11-01:
I was able to reproduce this by jumping to computer keyboard mode before letting the red line turn green,  If I waited for the line to turn green, I could jump back and forth fine. As, reported, all text was in the file.